ros/actionlib #35: waitForServer hangs when server name is incorrect

Tastalian, 2015-03-24:
Hello,

When a SimpleActionClient is constructed with the wrong action name, wait_for_server() hangs forever and there is no error message:

    client = actionlib.SimpleActionClient('wrong_action_name', DoDishesAction)
    client.wait_for_server()

I think there should be a check with an error message raised when the name is not found, as:
- I am not the only one facing this problem, see e.g. [this question on ROS answers](http://answers.ros.org/question/60307/simple-action-client-freezes-at-waitforserver-command-line-possibility/)
- it is common that the name declared in the SimpleActionServer() is not the same as that in the SimpleActionClient(), e.g. when the .launch file contains a <remap> of the topic name.

tfoote, 2015-03-24:
This isn't hanging. You have asked for the SimpleActionClient to wait until the expected resource becomes available with no timeout. 

If you would like it not to wait forever I suggest you pass it a timeout and process the return code. [Code API](http://docs.ros.org/indigo/api/actionlib/html/classactionlib_1_1SimpleActionClient.html#a0b00afd6df9d8c7bb1d02361ef8cf973) which will tell you that the server was not found and not wait forever. 

The server may or may not come from the same launch file. The code cannnot tell what your intent was, only how you configured it.

esteve, 2015-12-08:
I agree with @tfoote , this is not a bug in `waitForServer`, since it can't know whether you're asking for a resource that's invalid or that may become available in the future. Adding a timeout should prevent such behavior.

closed 2015-12-08 by esteve