ros/actionlib #38: Simple Action Server should expose setRejected

benaxelrod, 2015-04-13:
The server state transitions diagram (http://wiki.ros.org/actionlib/DetailedDescription) shows that when a goal comes in, it can either be accepted or rejected by the server.  When using the SimpleActionServer, your only choice is to accept the goal, even if it is invalid and you want to reject it.

tfoote, 2015-04-13:
The goal of the SimpleActionServer is to provide a simple solution which covers the simple use cases only, allowing an action server to be implemented [in a few lines of code](http://wiki.ros.org/actionlib#C.2B-.2B-_SimpleActionServer). It does not try to provide full coverage of the Action state space. If you want more complicated behaviours it's recommended to create your own action server implementation.

benaxelrod, 2015-04-13:
Understood.  However, SimpleActionServer already has the flexibility to allow users to do complicated things if they want by subscribing to the goal and preempt callbacks: http://wiki.ros.org/actionlib_tutorials/Tutorials/SimpleActionServer%28GoalCallbackMethod%29.  

I can certainly envision some advanced behaviors that one might want when a goal must be rejected, but I think the SimpleActionServer should do something simple like preempting the current goal.

I fail to see how adding this option in any way decreases the usefulness or simplicity of SimpleActionServer.

progtologist, 2016-11-30:
Resurrecting an old issue:
@benaxelrod actually the SimpleActionServer has very limited flexibility. If you examine closely the 
`template <class ActionSpec>
  void SimpleActionServer<ActionSpec>::goalCallback(GoalHandle goal)`
function you will see that the user can only do very few things with his custom goal callback. E.g. he cannot reject it and keep executing the previous goal as it's preemption status has been altered.
As @tfoote has suggested, you must implement your own ActionServer. My only -minor- complain is that there are no other Action Servers/Clients provided besides the Simple ones.