ros/actionlib #40: SimpleActionClient is tracking goal

jvgomez, 2015-11-06:
I have reached a point in my application in which I would like to have a SimpleActionClient member which allows me to know if the server is currently pursuing a goal. Something like `action_server.isTrackingGoal()`.

I guess is possible (I haven't checked in detail the SimpleActionClient implementation). Is there any reason not to do this? I think I could try out an implementation in case it is possible.

esteve, 2015-12-09:
`SimpleActionClient::getState()`'s documentation says this:

```
Possible States Are: PENDING, ACTIVE, RECALLED, REJECTED, PREEMPTED, ABORTED, SUCCEEDED, LOST.
   * \return The goal's state. Returns LOST if this SimpleActionClient isn't tracking a goal.
```

Could you try this? I think this will work, let me know if it's not what you intended.

```c++

SimpleActionClient simple_action_client(...);
bool isTrackingGoal = simple_action_client.getState() != SimpleClientGoalState::LOST;
```

Thanks!

jvgomez, 2016-03-11:
I have finally checked that method but there is a problem: if the server is not tracking any goal it returns LOST (as you mention) but also shows an error:

`ROS_ERROR("Trying to getState() when no goal is running. You are incorrectly using SimpleActionClient");`

Which is annoying, because in my opinion this does not mean that we are incorrectly using it. The only solution I can see is to store an additional flag to know whether any goal has been sent, and update it when the result is received.

Any further ideas?

luator, 2016-12-21:
I agree with jvgomez. Is there any reason to not simply remove the error message? I think this would be the easiest solution to the problem.

collinsc, 2018-02-23:
I also agree with jgomez. It's useful if you need to have multiple action clients in the same node, and juggle in between the two.

marcelinomalmeidan, 2018-10-23:
First of all, I would like to endorse this request! 
For those who are impatient to wait until the codebase is changed, there is an alternative.
You can actually see what is the current state of the action server by subscribing to the topic
`[action_name]/status`
This topic returns a message of the type `actionlib_msgs::GoalStatusArray`. In your code, you can check whether the action server is busy by having the following (C++):

`
		bool action_server_busy = false;
		for (uint i = 0; i < current_status.status_list.size(); i++) {
			if (current_status.status_list[i].status == current_status.status_list[i].ACTIVE) {
				action_server_busy = true;
			}
		}
`

doisyg, 2018-12-03:
Any progress on this topic?
PR here seems blocked : https://github.com/ros/actionlib/pull/97

PR #97 (2019-02-05): fix(actionlib): Remove `getState` error when no goal is running
As explained in #40 in many situations the developer needs to know if the action client is tracking a goal or not which is possible by the `getState` method of the `SimpleActionClient`. But the problem is, using this function while the action client didn't send any goal before, results to the following error:

> Trying to getState() when no goal is running. You are incorrectly using SimpleActionClient

This PR simply removes this ROS error since it's not really an incorrect usage of the `SimpleActionClient` (closes #40).
closed 2019-02-05 by mjcarroll