ros/actionlib #49: actionlib fails compilation with stricter GCC options

sytelus, 2016-05-21:
It looks like some of the actionlib uses unnecessary ";" which GCC correctly points out with stricter options. For example,

```
/opt/ros/indigo/include/actionlib/action_definition.h:58:2: error: extra ‘;’ [-Werror=pedantic]
/opt/ros/indigo/include/actionlib/destruction_guard.h:126:2: error: extra ‘;’ [-Werror=pedantic]
/opt/ros/indigo/include/actionlib_msgs/GoalID.h:177:36: error: extra ‘;’ [-Werror=pedantic]
/opt/ros/indigo/include/std_msgs/Header.h:188:36: error: extra ‘;’ [-Werror=pedantic]
/opt/ros/indigo/include/actionlib_msgs/GoalStatus.h:240:36: error: extra ‘;’ [-Werror=pedantic]

``` 

To reproduce these errors turn on checks in GCC using following line in CMakeLists.txt:

`set(CMAKE_CXX_FLAGS "-std=c++11 -Wall -Wextra  -Wstrict-aliasing -pedantic -fmax-errors=5 -Werror -Wunreachable-code ${CMAKE_CXX_FLAGS}")`

mikaelarguedas, 2016-05-24:
Indeed a lot of semicolons in the codebase are not necessary. Feel free to submit a pull request to address these in actionlib or any other package you want to compile with the `-pedantic` flag.
Thanks for pointing it out.

mikaelarguedas, 2016-08-25:
marking this with milestone `untargeted` given that we don't target these flags on ROS1 packages. We will still happily merge code cleaning pull requests

remod, 2019-04-30:
Addressed in https://github.com/ros/actionlib/pull/135.