ros/actionlib #56: Fix the silent fail when the robot stops short of a move goal

msy22, 2016-08-15:
I've been using actionlib to issue MoveBaseGoals to my robot for a while now. I've noticed that there are several instances where the robot gets close to the goal and stops, but doesn't properly transition to goal state 1 (SUCCEEDED). 

When this happens, there is effectively a silent fail because the robot refuses to move any further and actionlib doesn't report if it reached the goal or not. To resolve this I'd like to suggest adding the following few lines of code to the isGoalReached function in [goal_functions.cpp](http://docs.ros.org/jade/api/base_local_planner/html/goal__functions_8cpp_source.html) to warn the user if actionlib has stopped short of a move goal.

```C++
//Current code______________________________________________________________
double goal_x = goal_pose.getOrigin().getX();
double goal_y = goal_pose.getOrigin().getY();
double goal_th = tf::getYaw(goal_pose.getRotation());

//check to see if we've reached the goal position
if(getGoalPositionDistance(global_pose, goal_x, goal_y) <= xy_goal_tolerance) {
	//check to see if the goal orientation has been reached
	if(fabs(getGoalOrientationAngleDifference(global_pose, goal_th)) <= yaw_goal_tolerance) {
		//make sure that we're actually stopped before returning success
		if(stopped(base_odom, rot_stopped_vel, trans_stopped_vel)) 
		   return true;
	}
}  

//Suggested code______________________________________________________________
double goal_x = goal_pose.getOrigin().getX();
double goal_y = goal_pose.getOrigin().getY();
double goal_th = tf::getYaw(goal_pose.getRotation());

double distance_to_goal = getGoalPositionDistance(global_pose, goal_x, goal_y);
double angle_to_goal = fabs(getGoalOrientationAngleDifference(global_pose, goal_th))

//check to see if we've reached the goal position
if(distance_to_goal <= xy_goal_tolerance) {
	//check to see if the goal orientation has been reached
	if(angle_to_goal <= yaw_goal_tolerance) {
		//make sure that we're actually stopped before returning success
		if(stopped(base_odom, rot_stopped_vel, trans_stopped_vel)) {
		   return true;
		}
	}
}

//If we have stopped moving but didn't reach the goal, we need to report this failure to the user 
if(stopped(base_odom, rot_stopped_vel, trans_stopped_vel)) {
	//check to see if we failed to reach the goal position
	if(distance_to_goal > xy_goal_tolerance) {
		ROS_WARN("Robot stopped %d meters short of the goal position", fabs(distance_to_goal - xy_goal_tolerance)
	}
	//check to see if we failed to reach the goal orientation
	if(angle_to_goal > yaw_goal_tolerance) {
		ROS_WARN("Robot stopped %d radians short of the goal angle", fabs(angle_to_goal - yaw_goal_tolerance)
	}
}
``` 

I've read through a bit of the actionlib source code, but there may already be something like this that I haven't found yet, hence why this is just a suggestion. Or my problem may be totally unrelated, so I wanted to open this issue for discussion. I'm also new to Github so I don't know how to use pull requests or go about making changes like this on my own.

Any advice or constructive criticism would be appreciated.

tfoote, 2016-08-15:
This does not seem to be a problem with actionlib, but with the base_local_planner that it's still trying to reach the goal but is stuck. As you've suggested a change to the base_local_planner please file a ticket on the repo in which that file is hosted: https://github.com/ros-planning/navigation

As feedback on the suggested change, I would suggest formalizing the failure and detecting that the base_local_planner is stuck and abort the goal as unachievable in the case that it's actually stuck. Instead of just printing a warning. The warning is useful, if a human is watching it, but is not good for a supervisory process that might help get the robot out of the failure state.

msy22, 2016-08-15:
Thanks for the feedback @tfoote!

The thing is, when I review the bag files, the robot usually reaches the end of it's local path, and according to my own calculations it gets within the goal position and yaw tolerance, yet still doesn't correctly identify that the goal has been reached. I'll keep looking into the issue to determine if this is the real cause.

In the meantime, thanks for pointing out that I've opened this ticket in the wrong place. I'll close this one and open another under the base_local_planner.

closed 2016-08-15 by msy22