ros/actionlib #61: Simple action client/server warnings with Wextra

VictorLamoine, 2017-01-03:
If you compile a ROS node including the simple action client/server with `-Wextra` flags you will get a lot of compiler warnings.

How to reproduce
---
```
mkdir -p catkin_workspace/src
cd catkin_workspace/src
git clone https://github.com/frog0705/actionlib_tutorials
cd ..
catkin_make -DCMAKE_CXX_FLAGS="-Wall -Wextra"
```

```bash
[ 91%] Building CXX object actionlib_tutorials/CMakeFiles/fibonacci_client.dir/src/fibonacci_client.cpp.o
In file included from /opt/ros/kinetic/include/actionlib/client/client_helpers.h:47:0,
                 from /opt/ros/kinetic/include/actionlib/client/action_client.h:42,
                 from /opt/ros/kinetic/include/actionlib/client/simple_action_client.h:45,
                 from /home/victor/catkin_test/src/actionlib_tutorials/src/fibonacci_class_client.cpp:2:
/opt/ros/kinetic/include/actionlib/enclosure_deleter.h: In instantiation of ‘void actionlib::EnclosureDeleter<Enclosure>::operator()(Member*) [with Member = const actionlib_tutorials::FibonacciResult_<std::allocator<void> >; Enclosure = const actionlib_tutorials::FibonacciActionResult_<std::allocator<void> >]’:
/usr/include/boost/smart_ptr/detail/shared_count.hpp:171:14:   required from ‘boost::detail::shared_count::shared_count(P, D) [with P = const actionlib_tutorials::FibonacciResult_<std::allocator<void> >*; D = actionlib::EnclosureDeleter<const actionlib_tutorials::FibonacciActionResult_<std::allocator<void> > >]’
/usr/include/boost/smart_ptr/shared_ptr.hpp:358:76:   required from ‘boost::shared_ptr<T>::shared_ptr(Y*, D) [with Y = const actionlib_tutorials::FibonacciResult_<std::allocator<void> >; D = actionlib::EnclosureDeleter<const actionlib_tutorials::FibonacciActionResult_<std::allocator<void> > >; T = const actionlib_tutorials::FibonacciResult_<std::allocator<void> >]’
/opt/ros/kinetic/include/actionlib/client/comm_state_machine_imp.h:79:14:   required from ‘actionlib::CommStateMachine<ActionSpec>::ResultConstPtr actionlib::CommStateMachine<ActionSpec>::getResult() const [with ActionSpec = actionlib_tutorials::FibonacciAction_<std::allocator<void> >; actionlib::CommStateMachine<ActionSpec>::ResultConstPtr = boost::shared_ptr<const actionlib_tutorials::FibonacciResult_<std::allocator<void> > >; typename ActionSpec::_action_result_type::_result_type = actionlib_tutorials::FibonacciResult_<std::allocator<void> >]’
/opt/ros/kinetic/include/actionlib/client/client_goal_handle_imp.h:175:44:   required from ‘actionlib::ClientGoalHandle<ActionSpec>::ResultConstPtr actionlib::ClientGoalHandle<ActionSpec>::getResult() const [with ActionSpec = actionlib_tutorials::FibonacciAction_<std::allocator<void> >; actionlib::ClientGoalHandle<ActionSpec>::ResultConstPtr = boost::shared_ptr<const actionlib_tutorials::FibonacciResult_<std::allocator<void> > >; typename ActionSpec::_action_result_type::_result_type = actionlib_tutorials::FibonacciResult_<std::allocator<void> >]’
/opt/ros/kinetic/include/actionlib/client/simple_action_client.h:524:13:   required from ‘void actionlib::SimpleActionClient<ActionSpec>::handleTransition(actionlib::SimpleActionClient<ActionSpec>::GoalHandleT) [with ActionSpec = actionlib_tutorials::FibonacciAction_<std::allocator<void> >; actionlib::SimpleActionClient<ActionSpec>::GoalHandleT = actionlib::ClientGoalHandle<actionlib_tutorials::FibonacciAction_<std::allocator<void> > >]’
/opt/ros/kinetic/include/actionlib/client/simple_action_client.h:332:41:   required from ‘void actionlib::SimpleActionClient<ActionSpec>::sendGoal(const Goal&, actionlib::SimpleActionClient<ActionSpec>::SimpleDoneCallback, actionlib::SimpleActionClient<ActionSpec>::SimpleActiveCallback, actionlib::SimpleActionClient<ActionSpec>::SimpleFeedbackCallback) [with ActionSpec = actionlib_tutorials::FibonacciAction_<std::allocator<void> >; actionlib::SimpleActionClient<ActionSpec>::Goal = actionlib_tutorials::FibonacciGoal_<std::allocator<void> >; actionlib::SimpleActionClient<ActionSpec>::SimpleDoneCallback = boost::function<void(const actionlib::SimpleClientGoalState&, const boost::shared_ptr<const actionlib_tutorials::FibonacciResult_<std::allocator<void> > >&)>; typename ActionSpec::_action_result_type::_result_type = actionlib_tutorials::FibonacciResult_<std::allocator<void> >; actionlib::SimpleActionClient<ActionSpec>::SimpleActiveCallback = boost::function<void()>; actionlib::SimpleActionClient<ActionSpec>::SimpleFeedbackCallback = boost::function<void(const boost::shared_ptr<const actionlib_tutorials::FibonacciFeedback_<std::allocator<void> > >&)>; typename ActionSpec::_action_feedback_type::_feedback_type = actionlib_tutorials::FibonacciFeedback_<std::allocator<void> >]’
/home/victor/catkin_test/src/actionlib_tutorials/src/fibonacci_class_client.cpp:27:49:   required from here
/opt/ros/kinetic/include/actionlib/enclosure_deleter.h:60:52: warning: unused parameter ‘member_ptr’ [-Wunused-parameter]
     template<class Member> void operator()(Member* member_ptr){
                                                    ^~~~~~~~~~
In file included from /opt/ros/kinetic/include/actionlib/client/client_helpers.h:46:0,
                 from /opt/ros/kinetic/include/actionlib/client/action_client.h:42,
                 from /opt/ros/kinetic/include/actionlib/client/simple_action_client.h:45,
                 from /home/victor/catkin_test/src/actionlib_tutorials/src/fibonacci_class_client.cpp:2:
/opt/ros/kinetic/include/actionlib/managed_list.h: In instantiation of ‘void actionlib::ManagedList<T>::ElemDeleter::operator()(void*) [with T = boost::shared_ptr<actionlib::CommStateMachine<actionlib_tutorials::FibonacciAction_<std::allocator<void> > > >]’:
/usr/include/boost/smart_ptr/detail/shared_count.hpp:171:14:   required from ‘boost::detail::shared_count::shared_count(P, D) [with P = void*; D = actionlib::ManagedList<boost::shared_ptr<actionlib::CommStateMachine<actionlib_tutorials::FibonacciAction_<std::allocator<void> > > > >::ElemDeleter]’
/usr/include/boost/smart_ptr/shared_ptr.hpp:358:76:   required from ‘boost::shared_ptr<T>::shared_ptr(Y*, D) [with Y = void; D = actionlib::ManagedList<boost::shared_ptr<actionlib::CommStateMachine<actionlib_tutorials::FibonacciAction_<std::allocator<void> > > > >::ElemDeleter; T = void]’
/opt/ros/kinetic/include/actionlib/managed_list.h:214:58:   required from ‘actionlib::ManagedList<T>::Handle actionlib::ManagedList<T>::add(const T&, actionlib::ManagedList<T>::CustomDeleter, const boost::shared_ptr<actionlib::DestructionGuard>&) [with T = boost::shared_ptr<actionlib::CommStateMachine<actionlib_tutorials::FibonacciAction_<std::allocator<void> > > >; actionlib::ManagedList<T>::CustomDeleter = boost::function<void(actionlib::ManagedList<boost::shared_ptr<actionlib::CommStateMachine<actionlib_tutorials::FibonacciAction_<std::allocator<void> > > > >::iterator)>]’
/opt/ros/kinetic/include/actionlib/client/goal_manager_imp.h:74:138:   required from ‘actionlib::ClientGoalHandle<ActionSpec> actionlib::GoalManager<ActionSpec>::initGoal(const Goal&, actionlib::GoalManager<ActionSpec>::TransitionCallback, actionlib::GoalManager<ActionSpec>::FeedbackCallback) [with ActionSpec = actionlib_tutorials::FibonacciAction_<std::allocator<void> >; actionlib::GoalManager<ActionSpec>::Goal = actionlib_tutorials::FibonacciGoal_<std::allocator<void> >; actionlib::GoalManager<ActionSpec>::TransitionCallback = boost::function<void(actionlib::ClientGoalHandle<actionlib_tutorials::FibonacciAction_<std::allocator<void> > >)>; actionlib::GoalManager<ActionSpec>::FeedbackCallback = boost::function<void(actionlib::ClientGoalHandle<actionlib_tutorials::FibonacciAction_<std::allocator<void> > >, const boost::shared_ptr<const actionlib_tutorials::FibonacciFeedback_<std::allocator<void> > >&)>; typename ActionSpec::_action_feedback_type::_feedback_type = actionlib_tutorials::FibonacciFeedback_<std::allocator<void> >]’
/opt/ros/kinetic/include/actionlib/client/action_client.h:120:71:   required from ‘actionlib::ActionClient<ActionSpec>::GoalHandle actionlib::ActionClient<ActionSpec>::sendGoal(const Goal&, actionlib::ActionClient<ActionSpec>::TransitionCallback, actionlib::ActionClient<ActionSpec>::FeedbackCallback) [with ActionSpec = actionlib_tutorials::FibonacciAction_<std::allocator<void> >; actionlib::ActionClient<ActionSpec>::GoalHandle = actionlib::ClientGoalHandle<actionlib_tutorials::FibonacciAction_<std::allocator<void> > >; actionlib::ActionClient<ActionSpec>::Goal = actionlib_tutorials::FibonacciGoal_<std::allocator<void> >; actionlib::ActionClient<Action [... cut ...]
/opt/ros/kinetic/include/actionlib/client/simple_action_client.h:92:21:   required from ‘actionlib::SimpleActionClient<ActionSpec>::SimpleActionClient(const string&, bool) [with ActionSpec = actionlib_tutorials::FibonacciAction_<std::allocator<void> >; std::__cxx11::string = std::__cxx11::basic_string<char>]’
/home/victor/catkin_test/src/actionlib_tutorials/src/fibonacci_callback_client.cpp:34:30:   required from here
/opt/ros/kinetic/include/actionlib/enclosure_deleter.h:60:52: warning: unused parameter ‘member_ptr’ [-Wunused-parameter]
     template<class Member> void operator()(Member* member_ptr){
                                                    ^~~~~~~~~~
```

Not beautiful :sob:

mikaelarguedas, 2017-01-03:
duplicate of https://github.com/ros/actionlib/issues/49
marking this with milestone `untargeted` given that we don't target these flags on ROS1 packages. 
Feel free to submit a PR to fix these compiler warnings, we will happily review and merge it

mikaelarguedas, 2017-03-08:
closing this because it's a duplicate of #49 
PRs to make the code compile with stricter compiler options are always welcome!

closed 2017-03-08 by mikaelarguedas

mikaelarguedas, 2017-03-17:
fixed in #63 and #65, @VictorLamoine FYI

VictorLamoine, 2017-03-17:
Thanks for the reminder :+1: