ros/actionlib #67: ACtionlib Handle assertion killed program

pbeeson, 2017-02-20:
I have a pretty intensive robotic application that we are running through reliability tests before software goes into production in June.  After about a week of running fine, one our actionlib clients died due to a low-level assertion in the ManagedList class.

```
affordance_template_server_node: /home/pbeeson/ros/vehtek/src/actionlib/include/actionlib/managed_list.h:171: bool actionlib::ManagedList<T>::Handle::operator==(const actionlib::ManagedList<T>::Handle&) const [with T = boost::shared_ptr<actionlib::CommStateMachine<affordance_template_msgs::PlanAction_<std::allocator<void> > > >]: Assertion `valid_' failed.
```

pbeeson, 2017-02-20:
Could it be that on line 182 of managed_list.h, that the `valid_(true)` (the only place valid_ is ever set to true) should come before the initialization of `it_` and `handle_tracker_`?

pbeeson, 2017-02-20:
So I don't know this code well, but it looks like valid_ is being used to set whtehr a handle is valid or not (duh) but the the handle is still in scope somewhere, so I'm guessing there is a race condition where a new handle gets genreated and something is comparing it to any old handles, and the old one hasn't been completely destroyed yet, but has been marked valid, and that causes this assertion to fail.

pbeeson, 2017-02-21:
My current "fix" for this is to go through the client code and replace all asserts with ROS_ERROR messages, and return default values where appropriate.

[Temp fix](https://github.com/pbeeson/actionlib/commit/d761eadb7bb2fbbe3bd23e41195cb106135dcff8)

pbeeson, 2017-02-28:
Anyone have thoughts on this?

pbeeson, 2017-03-03:
After 5 days of running, I haven't encountered the error message I replaced the assertions with, but I am seeing this:
`Got a callback on a goalHandle that we're not tracking.  This is an internal SimpleActionClient/ActionClient bug.  This could also be a GoalID collision`
My feeling is that this symptom is from the same underlying problem, in that goals can be deleted or partially deleted in the client from a Done/Finished message before some later feedback messages get processed about the now missing goal in the actionlib client.

mikaelarguedas, 2017-07-18:
addressed in #83. Note the code still needs to be compiled in Release/RelWithDebInfo build type for the asserts to be removed

closed 2017-07-18 by mikaelarguedas