ros/actionlib #91: SimpleActionServer<>::shutdown() deletes execute_thread_ twice

lsolanka, 2017-12-11:
Hi,

This code seems to have some duplication:
```C++
template<class ActionSpec>
void SimpleActionServer<ActionSpec>::shutdown()
{
  if (execute_callback_) {
    {
      {
        boost::mutex::scoped_lock terminate_lock(terminate_mutex_);
        need_to_terminate_ = true;
      }

      assert(execute_thread_);
      if (execute_thread_) {
        execute_thread_->join();
        delete execute_thread_;
        execute_thread_ = NULL;
      }
    }

    assert(execute_thread_);
    execute_thread_->join();
    delete execute_thread_;
    execute_thread_ = NULL;
  }
}
```
In debug builds the second assertion fails always.

It looks like it was introduced in commit bac04a3989c1436d2b9f8300c82ce3cbab0ceee9?

Lukas

mikaelarguedas, 2017-12-11:
Thanks @lsolanka for reporting.
This is indeed wrong, the scoping is weird too,  this must have come from a rebasing issue when #72 was merged. See #92 for a proposed fix

closed 2017-12-11 by mikaelarguedas

v-lopez, 2018-01-17:
@mikaelarguedas Any ETA for the release on this? On the last Kinetic sync (January 9th) this was not included.

mikaelarguedas, 2018-01-18:
@v-lopez, thanks for pointing it out. The release has been tagged and will be pushed during the next kinetic and lunar sync

v-lopez, 2018-01-18:
@miguelsdc Thanks! Take care

pbeeson, 2018-01-25:
1.11.12 does not seem to have be released into the build farm yet.  When is the next kinetic sync scheduled? This bug causes actionlibs segfault on shutdown and is having serious impacts on our customers, as our code creates and destroys threads with new actionlibs every so often.  It's going to be a pain to remote into dozens of machines and catkin clean, clone actionlib, and catkin build for our entire software system, so any .deb at all (even in the shadow repo) would be greatly appreciated.

mikaelarguedas, 2018-01-25:
>  It's going to be a pain to remote into dozens of machines and catkin clean, clone actionlib, and catkin build for our entire software system

Totally understandable, sorry for the inconvenience, a deb will be available shortly, more details below.

1.11.12 has been released in [Kinetic](https://github.com/ros/rosdistro/pull/16821) and [Lunar](https://github.com/ros/rosdistro/pull/16822) and has been built. Once the downstream packages are done rebuilding it will be synced to the shadow repository.

Kinetic still has some serious regressions delaying the sync for now. Lunar is all green so we can do an express sync if needed (but apparently that's kinetic that's of interest here).

pbeeson, 2018-01-25:
Great thanks!

mikaelarguedas, 2018-01-25:
Talking with @tfoote, it should also be possible to sync Kinetic soon as the regressing packages have been reverted to an earlier working version

mikaelarguedas, 2018-01-31:
@pbeeson FYI: actionlib 1.11.12 has been synced on [Lunar](https://discourse.ros.org/t/new-packages-for-lunar-2018-01-30/3848) and [Kinetic](https://discourse.ros.org/t/new-packages-for-kinetic-2018-01-30/3849)

pbeeson, 2018-01-31:
I noticed.  I had been wget-ing the .deb from the Shadow Repo last week, but this makes life easier moving forward.  Thanks for prioritizing this.

PR #92 (2017-12-11): fix uncrustify mixup
follow up of #72 
fixes #91