ros/actionlib #129: Actionlib pulls GUI dependencies

ianjfrosst, 2019-03-20:
When installing ROS, the `ros-base` set of packages is documented as having "no GUI tools", making it ideal for lightweight headless systems. Actionlib, as part of the core ROS stack, is included as part of this configuration. However, Actionlib includes two GUI tools, written in Python (`axclient.py` and `axserver.py`), and as of #79, includes their dependency `python-wxtools`, which pulls in nearly 100 additional packages, including GTK and `x11-common`.

ianjfrosst, 2019-03-25:
Bump @mjcarroll

mjcarroll, 2019-03-25:
@ianjfrosst Thanks for the heads up.  I think that the correct thing to do here would be to move those GUI-dependent scripts into their own package to keep them out of ros-base.

If you have time/comfort level to make those modifications, I would happily review, otherwise I can put it in my backlog, but not sure when I can get to it.

gavanderhoorn, 2019-03-31:
@mjcarroll: seeing as this repository currently hosts a single package at its root, would you rather this repository's layout is changed (ie: push everything down into a directory for the package), or should a new repository be created?

ianjfrosst, 2019-03-31:
I've made those modifications (created a new package `actionlib_tools`), but was stuck on @gavanderhoorn's point. Currently, I have them in a separate repository, which normally I'm not a fan of (breaking up git history), but in this case, those tools have only ~7 commits that touch them.

gavanderhoorn, 2019-04-01:
Pushing everything in this repository down into an `actionlib` package and then adding your `actionlib_tools` probably makes the most sense: the packages are closely related (as in: changes to `actionlib` will probably require changes to `actionlib_tools` as well) deal with the same topic and versioning and releasing them together would then make sense as well.

But I'm not a/the maintainer here, so let's see what @mjcarroll prefers.

mjcarroll, 2019-04-01:
I'm okay with modifying the layout of the repository for the same reasons that @gavanderhoorn has stated above.

mikepurvis, 2019-04-04:
Yes please to this change— it's also a hassle because the story for `python-wxtools` and Python 3 is a little murky right now, so dropping the dependency is a good step forward on that. See:

https://stackoverflow.com/questions/26251030/how-to-use-wxpython-for-python-3
https://pypi.org/project/wxPython/

Neither of these two packages are available upstream in a `python3-` version as of Ubuntu disco:

https://packages.ubuntu.com/disco/python-wxtools
https://packages.ubuntu.com/disco/python-wxgtk3.0

ianjfrosst, 2019-04-06:
#133 implements these changes.

ianjfrosst, 2019-12-18:
#157 implements these changes but better.

130s, 2023-06-14:
Moving GUI dependency to a separate pkg in https://github.com/ros/actionlib/pull/152 might have resolved this?

ianjfrosst, 2023-06-15:
Probably! I don't have an environment to test it right now, but if that got merged then this issue is likely resolved.

closed 2025-05-31 by sloretz