ros/actionlib #153: Noetic Python 2 dependency on wxtools

sloretz, 2019-10-22:
It looks like the [noetic source entry for `actionlib`](https://github.com/ros/rosdistro/blob/1b96f7c4f230e878f9901a6cc3465af9cbca22fc/noetic/distribution.yaml#L13-L19) has a `package.xml` that depends on the Python 2 rosdep key `python-wxtools`.

https://github.com/ros/actionlib/blob/c9d442a86556a866ce2afceb3e055a9d0f235e89/actionlib_tools/package.xml#L24

I don't think there is an equivalent rosdep key for Python 3. Can the dependency be eliminated?

130s, 2020-01-16:
Not entirely sure but https://stackoverflow.com/a/20614692/577001 says wxpython 4, which seems to be already available via `pip`, is compatible for both Python 2 and 3. Today I see [python-wxtools for Ubuntu Focal is 3.0.2](https://packages.ubuntu.com/focal/python-wxtools). If I'm correctly seeing the situation, one way to go may be to make a dep rule for focal for pip?

Obviously "cleaner" option might be to update `ax{server, client}.py` to replace `wx` with something else (in a similar way done for core GUI tools using `qt` when `rqt` was introduced). Looks like those are very light weight GUI so I imagine the coversion will be not hard.

sloretz, 2020-01-16:
> If I'm correctly seeing the situation, one way to go may be to make a dep rule for focal for pip?

I don't think debian packages can depend on pip packages though. The [places where wx is imported](https://github.com/ros/actionlib/blob/c89e440519eeb88a5005065a32330655c9450b8e/tools/axclient.py#L46) would fail if the only thing someone did was install `ros-noetic-actionlib`.

130s, 2020-01-16:
Ah you're right.

Btw not a solution to this issue, but I found https://github.com/ros/actionlib/issues/129 is related.

sloretz, 2020-04-23:
@mjcarroll friendly ping. It looks like `actionlib_tools` is released in Noetic with a Python 2 depedency. Mind removing it?

[https://github.com/ros-gbp/actionlib-release/blob/debian/ros-noetic-actionlib-tools_1.13.0-1_focal/package.xml#L24](https://github.com/ros-gbp/actionlib-release/blob/c33e80ff9c9544a2b1884af3c5f2f510e0a3e4e6/package.xml#L24)

mjcarroll, 2020-04-23:
I'll take a look @sloretz, thanks for the reminder.

sloretz, 2020-06-05:
@mjcarroll Might be able to use https://packages.debian.org/buster/python3-wxgtk4.0, but will need a rosdep key