ros/actionlib #154: Segmentation Fault when I run client node

scopeccsky, 2019-11-04:
ROS-melodic
ubuntu 18.04

I am learning actionlib, and I copy code from wiki actionlib and catkin_make success, but when I run client, the fault always at "Client client("do_dishes", true); // true -> don't need ros::spin()"

This is the information from gdb:

#0  0x00007ffff6a99f13 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#1  0x00007ffff6fa969c in boost::detail::add_new_tss_node(void const*, boost::shared_ptr<boost::detail::tss_cleanup_function>, void*) () from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.65.1
#2  0x00007ffff6faa5cc in boost::detail::set_tss_data(void const*, boost::shared_ptr<boost::detail::tss_cleanup_function>, void*, bool) () from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.65.1
#3  0x00007ffff7905216 in ros::CallbackQueue::setupTLS() () from /opt/ros/melodic/lib/libroscpp.so
#4  0x00007ffff7907f3f in ros::CallbackQueue::callAvailable(ros::WallDuration) () from /opt/ros/melodic/lib/libroscpp.so
#5  0x00005555555ec35b in actionlib::SimpleActionClient<actionlib_test::FibonacciAction_<std::allocator<void> > >::spinThread (this=0x7fffffffd580) at /opt/ros/melodic/include/actionlib/client/simple_action_client.h:297
#6  0x00005555556114d3 in boost::_mfi::mf0<void, actionlib::SimpleActionClient<actionlib_test::FibonacciAction_<std::allocator<void> > > >::operator() (this=0x555555864e38, p=0x7fffffffd580)
    at /usr/local/include/boost/bind/mem_fn_template.hpp:49
#7  0x0000555555610b5b in boost::_bi::list1<boost::_bi::value<actionlib::SimpleActionClient<actionlib_test::FibonacciAction_<std::allocator<void> > >*> >::operator()<boost::_mfi::mf0<void, actionlib::SimpleActionClient<actionlib_test::FibonacciAction_<std::allocator<void> > > >, boost::_bi::list0> (this=0x555555864e48, f=..., a=...) at /usr/local/include/boost/bind/bind.hpp:259
---Type <return> to continue, or q <return> to quit---
#8  0x000055555560ff1e in boost::_bi::bind_t<void, boost::_mfi::mf0<void, actionlib::SimpleActionClient<actionlib_test::FibonacciAction_<std::allocator<void> > > >, boost::_bi::list1<boost::_bi::value<actionlib::SimpleActionClient<actionlib_test::FibonacciAction_<std::allocator<void> > >*> > >::operator() (this=0x555555864e38) at /usr/local/include/boost/bind/bind.hpp:1294
#9  0x000055555560e816 in boost::detail::thread_data<boost::_bi::bind_t<void, boost::_mfi::mf0<void, actionlib::SimpleActionClient<actionlib_test::FibonacciAction_<std::allocator<void> > > >, boost::_bi::list1<boost::_bi::value<actionlib::SimpleActionClient<actionlib_test::FibonacciAction_<std::allocator<void> > >*> > > >::run (this=0x555555864d00) at /usr/local/include/boost/thread/detail/thread.hpp:120
#10 0x00007ffff6fa9bcd in ?? () from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.65.1
#11 0x00007ffff6d806db in start_thread (arg=0x7fffeaffd700) at pthread_create.c:463
#12 0x00007ffff650888f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

Thanks.

robodyne-robotics, 2020-04-11:
I'm having the same issue and I have the same Ubuntu version, too.

fujitatomoya, 2020-04-14:
@scopeccsky @robodyne-robotics 

are you referring to [actionlib/Tutorials](http://wiki.ros.org/actionlib/Tutorials)?
can you try to use [actionlib_tutorials](https://github.com/ros/common_tutorials/tree/indigo-devel/actionlib_tutorials) instead?

I think that wiki is to explain and share insights, maybe it is better to use actual tutorial code.