ros/actionlib #167: [noetic] Where is axclient.py

doisyg, 2020-05-27:
I use quite a lot  `rosrun actionlib axclient.py` but I don't find it under the noetic release

doisyg, 2020-05-27:
Related to the python 2 wxtools dependencies... https://github.com/ros/actionlib/issues/153
axclient.py and scripts/axserver.py  are not installed anymore
https://github.com/ros/actionlib/blob/b58e1228af606d449f988f4609f176bb3353b357/actionlib_tools/CMakeLists.txt#L8-L14

mikaelarguedas, 2020-06-19:
+1 this is a valuable tool with no real replacement available in Noetic.
Commenting to voice my support for its' Python 3 conversion and return in the debs :)

mikaelarguedas, 2020-08-16:
@doisyg @reinzor @BG-Keith I submitted a Noetic compatible version at https://github.com/ros/actionlib/pull/169 if you guys want to give it a try

reinzor, 2020-08-17:
Thanks!

[image]

Seems to work nicely here!

hcostelha, 2020-10-12:
I have testes this with the current Ubuntu available repository package and is working fine, so I guess this issue can be closed.

closed 2021-02-19 by doisyg

rickstaa, 2021-05-21:
For future reference, the tool is now found under the `actionlib_tools` package and can be started using the:

```bash
rosrun actionlib_tools axclient.py
```

command.

rickstaa, 2021-05-21:
For future reference, the `axclient.py` tool has been moved to the `ros-noetic-actionlib-tools` package.

130s, 2023-05-05:
> I could not find the official wiki for this subpackage.

Just made it http://wiki.ros.org/actionlib_tools, for ROS1. 

Don't know where it went in ROS2 world :thinking: