ros/actionlib #192: noetic install broken

Russ76, 2021-07-31:
sudo apt-get install ros-noetic-actionlib
Reading package lists... Done
Building dependency tree       
Reading state information... Done
ros-noetic-actionlib is already the newest version (1.13.2-1focal.20210423.233703).
ros-noetic-actionlib set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 59 not upgraded.
2 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
Setting up amdgpu-dkms (1:5.6.0.15-1098277) ...
Removing old amdgpu-5.6.0.15-1098277 DKMS files...

------------------------------
Deleting module version: 5.6.0.15-1098277
completely from the DKMS tree.
------------------------------
Done.
Loading new amdgpu-5.6.0.15-1098277 DKMS files...
Building for 5.8.0-63-generic
Building for architecture x86_64
Building initial module for 5.8.0-63-generic
ERROR: Cannot create report: [Errno 17] File exists: '/var/crash/amdgpu-dkms-firmware.0.crash'
Error! Bad return status for module build on kernel: 5.8.0-63-generic (x86_64)
Consult /var/lib/dkms/amdgpu/5.6.0.15-1098277/build/make.log for more information.
dpkg: error processing package amdgpu-dkms (--configure):
 installed amdgpu-dkms package post-installation script subprocess returned error exit status 10
No apport report written because the error message indicates its a followup error from a previous failure.
                          dpkg: dependency problems prevent configuration of amdgpu:
 amdgpu depends on amdgpu-dkms (= 1:5.6.0.15-1098277); however:
  Package amdgpu-dkms is not configured yet.

dpkg: error processing package amdgpu (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 amdgpu-dkms
 amdgpu
E: Sub-process /usr/bin/dpkg returned an error code (1)

Did my machine try to make a new binary for this module? I don't understand these error messages.
Noetic, Ubuntu 20.04, HP Pavilion laptop, Ryzen 7, new, clean install

gavanderhoorn, 2021-08-17:
Cross-post of [Actionlib didn't install](https://answers.ros.org/question/383604).

tfoote, 2021-08-17:
This is not a problem with actionlib installing. You already have the latest version on your system `ros-noetic-actionlib is already the newest version (1.13.2-1focal.20210423.233703).
ros-noetic-actionlib set to manually installed.`

Thanks to @gavanderhoorn for the cross link closing this here as he's already answered you on the answers post. You need to look at the broken installation of amdgpu-dkms.

closed 2021-08-17 by tfoote