ros/actionlib #213: Lost feedback on python actionclient

Doomerdinger, 2023-03-02:
If using the python action client, sometimes the last (1 or more) feedback message(s) can get lost if the goal is marked as completed before the feedback callback is triggered.

This appears to be an intentional decision as seen [here](https://github.com/ros/actionlib/blob/23acb6e7364dda9380f823e03284536574764c6a/actionlib/src/actionlib/action_client.py#L416), where the feedback callback is skipped if the commstate is done.

However, I do not see this issue when using the c++ client. The same code on the c++ side can be seen [here](https://github.com/ros/actionlib/blob/23acb6e7364dda9380f823e03284536574764c6a/actionlib/include/actionlib/client/comm_state_machine_imp.h#L127), which does not appear to check if the goal is completed or not.

This causes issues on the python side if the feedback is significant and we want to ensure we receive all feedback the server sends.

If my understanding of the c++ functionally is correct, this seems to be a mistake on the python side.
If I can get some consensus here I will happily open a PR.

Timple, 2023-03-03:
Interesting that there is a discrepancy between c++ and python...

I do wan't to add a concern here. We send a lot of consecutive action goals. What I certainly would see a regression if feedback from a previous goal ended up while a new goal was send out.

Our feedback messages contain a progress. Now imagine we send out a new goal, but a 99.9% feedback message arrives from a previous goal. We would draw wrong conclusions from this...

On the other hand, if no new goal is active we don't mind receiving it.

Personally I always saw action feedback like I see (non-latched) topics. Information might be missed. Therefor we put all important info in the result of the action. But of course this does not have to hold for everyone.

Doomerdinger, 2023-03-06:
That's a good point.

Could potentially account for that using the `goal_id`.

Either way I'd just like it to be consistent across the two platforms which I do not _think_ it is at the moment.

closed 2025-05-31 by sloretz